moggieuk/ERCF-Software-V3 #1: Stall guard testing ...

nigelpjames, 2022-11-30:
Hi, Ive been using stall guard with easy brd and the mod whi-tw posted on pull request (https://github.com/EtteGit/EnragedRabbitProject/pull/150) for some time with good success and have subsequently removed my hall affect sensor.   Its doable with easy brd provided sensorless selector homing is disabled and manually jumper the 2209 diag pin. With the original patch, Klipper would occasionally cancel the print with "!! No trigger on manual_stepper gear_stepper after full movement" if stall guard failed to home & triggered the limit.  Changing homing_move(self, movepos, speed, probe_pos=False, triggered=True, check_triggered=false) solved this problem and haven't had a repeat issue. 
I also added a short 4mm move to detention the filament path much like your 2mm delay before releasing the servo.

I'm keen to move to your driver but couldnt get stall guard to work reliably and have reverted to my previous setup.  Sensorless homing without stall guard was reliable but noisier than expected even after dropping currents to 10%. Sync load was also unexpectedly noisy and found I needed to disable it but ran out of time to debug.  

Happy to help debug stall guard with your driver if you arent already working on this with others as zero grinding and practically silent feed through to the extruder is possible with it.

nigelpjames, 2022-11-30:
I presume the comment you have in your code referring to Easy Brd and stall guard not being compatible has something to do with the method you are using to drive the stepper.  I have the 2209 diag pin jumpered and the following code works with stall guard.  When I enable stall guard in the v3 fork, it activates the selector rather than gear motor.

    def _home_to_extruder_with_stallguard(self, max_length):
        self._servo_down()
        self.toolhead.dwell(0.2)
        self.toolhead.wait_moves()

        self._log_debug("Homing to extruder with stallguard, up to %.1fmm" % max_length)

        self.gear_stepper.do_set_position(0.)

        pre_move_position = self._counter.get_distance()
        #self.gear_stepper.do_homing_move(max_length, 5, self.gear_stepper_accel, True, True)
        **self.gear_stepper.do_homing_move(max_length, 5, self.gear_stepper_accel, True, False)**
        self.toolhead.dwell(0.2)
        self.toolhead.wait_moves()
        post_move_position = self._counter.get_distance()
        distance_moved = post_move_position - pre_move_position
        if distance_moved >= max_length:
            self._log_info("Failed to reach extruder after moving %.1fmm, pausing" % distance_moved)
            self._pause()
        else:
            self._log_debug("Extruder reached after %.1fmm" % distance_moved)
            return post_move_position

moggieuk, 2022-12-07:
HI, I think we have been chatting in discord, but whats your status of stallguard for filament homing now with the little update I made in the last 1.1.5 release?

moggieuk, 2022-12-16:
I believe you verified that stallguard is now working for sensorless homing so I'm going to close this issue.

closed 2022-12-16 by moggieuk