moggieuk/ERCF-Software-V3 #2: Failure on toolswap without print running still issues a PAUSE

spikeygg, 2022-12-03:
When testing tool swaps (issuing `T0`, for example) WITHOUT a print running, the software still issues a PAUSE command on failure. I don't think it should, when the printer becomes paused without a running print KlipperScreen goes into a weird state and cannot be recovered from its GUI (separate issue, that was brought to light by this one).

To recover the printer's state, I have to get to the console and issue ERCF_UNLOCK and RESUME.

moggieuk, 2022-12-07:
Thanks.  There was an error in endless spool.  This should be fixed in v1.1.5-4 release

spikeygg, 2022-12-07:
Actually, this may be a non-issue after your description. I figured that klipper would know if it's printing or not with a boolean state. I opened a [similar ticket in the KlipperScreen github](https://github.com/jordanruthe/KlipperScreen/issues/832) and it has been resolved with a modification one-liner. I've already pulled this change in and tested it out. It resolved the weird locked screen, otherwise I would get you a screenshot.

Maybe a PAUSE issued when the printer isn't printing isn't a problem. I did include a CLEAR_PAUSE in my PRINT_START a long time ago when I first started using the ERCF because the ERCF at that time would issue several PAUSEs and something would cause klipper to get confused and not actually PAUSE while also claiming that the printer was already paused. I found that the CLEAR_PAUSE command before starting the print usually cleared out that issue. The CLEAR_PAUSE at PRINT_START probably would keep any left-over PAUSE from causing issues.

closed 2022-12-07 by spikeygg