moggieuk/ERCF-Software-V3 #5: Internal error on command:"ERCF_ENCODER_RUNOUT"

adolfotregosa, 2022-12-07:
I dunno what more to do. Everything works fine, I can change tools succesfully but i'm trying out endless spool but It always crashes with Internal error on command:"ERCF_ENCODER_RUNOUT".

Help ?

11:51
Internal error on command:"ERCF_ENCODER_RUNOUT"
11:51
Klipper state: Shutdown
11:51
Unloaded 955.6mm of filament
11:51
- DEBUG: Setting servo to up angle: 30
11:50
ERCF [T0] <.... [encoder] .............. [extruder] .... [sensor] .... [nozzle] UNLOADED (@955.6 mm)
11:50
- - TRACE: Final parking. Stepper: 'gear' moved -10.7mm, encoder measured 0.0mm (delta 10.7mm). Counter: @955.6mm
11:50
ERCF [T0] <<<.. [encoder] .............. [extruder] .... [sensor] .... [nozzle] (@955.6 mm)
11:50
- - TRACE: Unloading step #3 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 2.7mm (delta 12.3mm). Counter: @955.6mm
11:50
- - TRACE: Unloading step #2 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 14.3mm (delta 0.7mm). Counter: @952.9mm
11:50
- - TRACE: Unloading step #1 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 15.0mm (delta 0.0mm). Counter: @938.6mm
11:50
- DEBUG: Slow unload of the encoder
11:50
ERCF [T0] <<<<< [encoder] <<<........... [extruder] .... [sensor] .... [nozzle] (@923.6 mm)
11:50
ERCF [T0] <<<<< [encoder] <<<<<<<<...... [extruder] .... [sensor] .... [nozzle] (@923.6 mm)
11:50
- - TRACE: Course unloading move #1 from bowden. Stepper: 'gear' moved -884.9mm, encoder measured 886.9mm (delta -2.0mm). Counter: @923.6mm
11:50
- - TRACE: Sync unload. Stepper: 'both' moved -15.0mm, encoder measured 15.0mm (delta 0.0mm). Counter: @36.8mm
11:50
- DEBUG: Moving the gear and extruder motors in sync for -15.0mm
11:50
- DEBUG: Setting servo to down angle: 140
11:50
- DEBUG: Unloading bowden tube
11:50
ERCF [T0] <<<<< [encoder] <<<<<<<<<<<<<< [extruder] .... [sensor] .... [nozzle] (@14.3 mm)
11:50
- DEBUG: Filament should be out of extruder
11:50
- - TRACE: Last sanity move. Stepper: 'extruder' moved -35.0mm, encoder measured 0.0mm (delta 35.0mm). Counter: @14.3mm
11:50
- DEBUG: Toolhead sensor reached after 1 moves
11:50
ERCF [T0] <<<<< [encoder] <<<<<<<<<<<<<< [extruder] <<<| [sensor] .... [nozzle] (@14.3 mm)
11:50
- - TRACE: Step #1:. Stepper: 'extruder' moved -15.0mm, encoder measured 12.9mm (delta 2.1mm). Counter: @14.3mm
11:50
- DEBUG: Trying to exit the extruder, up to 95.0mm in 15.0mm steps
11:50
- DEBUG: Extracting filament from extruder
11:50
ERCF [T0] <<<<< [encoder] <<<<<<<<<<<<<< [extruder] <<<< [sensor] <<<< [nozzle] LOADED (@0.0 mm)
11:50
Unloading filament...
11:50
- - TRACE: Disable encoder sensor
11:50
- DEBUG: Unloading tool T0
11:50
- - TRACE: After tip formation, encoder moved 54.45
11:50
pressure_advance: 0.000000
pressure_advance_smooth_time: 0.020000
11:50
Forming tip...
11:50
Remapping T0 to gate #1
11:50
EndlessSpool checking for additional spools in group 1...
11:50
A runout has been detected
11:50
- DEBUG: Setting servo to up angle: 30
11:50
- - TRACE: After buzzing gear motor, encoder moved 0.00
11:50
- DEBUG: Setting servo to down angle: 140
11:50
- - TRACE: Disable encoder sensor
11:50
- DEBUG: Checking if this is a clog or a runout...
11:50
Issue on tool T0

moggieuk, 2022-12-07:
Do you have the klippy.log when this happened?

There could be a bug in the software or something is not configured correctly...   I want to find / verify the issue but you should probably also increase the 'ercf_calib_clog_length' value in your ercf_vars.cfg file and restart Klipper.   I attempt to guess at a sensible value but it appears it might not be quite enough.

moggieuk, 2022-12-07:
I believe the bug in endless spool is fixed in v1.1.5-4 update.

moggieuk, 2022-12-08:
Closing because "Internal Error" issue is fixed and the false triggering of runout just needs some adjustment of the ercf_calib_clog_length.

closed 2022-12-08 by moggieuk