moggieuk/ERCF-Software-V3 #6: Hi I have a suggestion for multicolor printing

Maeians, 2022-12-20:
Is it possible to install a servo on the extruder sensor side?
I can redesign and add extruder sensor + material cut sensor
The purpose is to cut the material
Similar to the extruder in the X1 Carbon Combo

[image]

Maeians, 2022-12-20:
video link
[https://www.bilibili.com/video/BV1aG4y1H75U/?spm_id_from=333.880.my_history.page.click&vd_source=d48c8938a38a9051ebe0d653cad672a6](url)

located in 03:20-03:56

Durahl, 2022-12-25:
1. Doesn't necessarily need a Servo **on the Extruder** as it could be remotely triggered via a Bowden Cable like a Bicycle Brake.
2. Doesn't **need a Servo** if you design for the Mechanism to be triggered by contacting the moving Extruder against a stationary part of the Gantry.
3. **Doesn't need to be** to begin with as the amount of material that could have been extracted from but now left in the Hot End is HUGE and something I hear all reviewers complain about when discussing the X1 Carbo as all of that material then needs to be purged out first.

rwijnhov, 2022-12-28:
filament cutting is a really nice feature. It works great on the mmu2s in the end. And proven verry useful.

moggieuk, 2022-12-30:
This would be a very interesting addition.  It would be very easy to integrate into Happy Hare especially if a servo.  If a mod is built I certainly can add support for it...

closed 2023-02-02 by moggieuk