moggieuk/ERCF-Software-V3 #8: ERCF synchronized with extruder motor

RickDnamps, 2022-12-30:
Do you think it could be possible yo add the opyion to use the gear ercf motor simultaneously with the Extruder to do a push pull cobfiguration to help wiyh the long bowden tube??? I have comprnsated that eith a second extruder but it's not the best.....

Thanks

moggieuk, 2022-12-30:
Theoretically possible but unfortunately very difficult.  To do so it would need some major changes to Klipper but also keeping the two motors in sync would require some additional sensor and feedback logic. Even a slight difference between length of filament pushed by each motor would create a problem...

Rather than keep then exactly in sync, one idea would be to use the gear motor to keep slack in the filament for the extruder. This would require some sort of sensor and feedback loop and again integration into klipper but perhaps a little less complicated.

moggieuk, 2023-02-02:
Closing as not possible / practical without additional sensors.

closed 2023-02-02 by moggieuk