moggieuk/ERCF-Software-V3 #12: Typo in ercf.py: home_poaition_to_nozzle_speed

spikeygg, 2023-01-01:
BeOddTodd found a typo in the ercf.py file:
https://github.com/moggieuk/ERCF-Software-V3/blob/84d86d7007c2073b8de4f589353132cd0e03d6d2/extras/ercf.py#L1595

I think it should be `home_position_to_nozzle_speed`, not `home_poaition_to_nozzle_speed`

nigelpjames, 2023-01-02:
self.nozzle_load_speed perhaps

Danielstc91, 2023-01-02:
So I changed to home_position_nozzle_speed
Which then got the error 
AttributeError: 'Ercf' object has no attribute 'home_position_to_nozzle_speed'

I removed the  "_speed" part of the name (as it is not refereced anywhere else in the code)
And that got rid of those issues. 

Now i see 

`Traceback (most recent call last):

  File "/home/pi/klipper/klippy/extras/ercf.py", line 404, in handle_connect
    self.toolhead_sensor = self.printer.lookup_object("filament_switch_sensor toolhead_sensor")
  File "/home/pi/klipper/klippy/klippy.py", line 97, in lookup_object
    raise self.config_error("Unknown config object '%s'" % (name,))
configparser.Error: Unknown config object 'filament_switch_sensor toolhead_sensor'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 446, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
BlockingIOError: [Errno 11] Resource temporarily unavailable
- DEBUG: ercf_log=/home/pi/klipper_logs/ercf.log'

Danielstc91, 2023-01-02:
> self.nozzle_load_speed perhaps

This also works to fix the original issue but there are still issues afterwards.
This leaves the same errors as i posted above

moggieuk, 2023-01-02:
Thanks all.  I'm fixing the speed setting (I accidentally notice because I use a "sync" move when loading - this error occurs occurs if not syncing.

@Danielstc91 The exception you are seeing... can you tell me how this happens? When initially starting up? Or when "klipper restart" or other?    It works completely as expected for me: detection that you have no toolhead sensor and forcing home_to_extruder...

Danielstc91, 2023-01-02:
@moggieuk This happens after i call T0 or T1 or any tool. 
It will run the gear motor to the extruder gear. then klipper shuts down and gives an internal error for T0 command.

moggieuk, 2023-01-02:
I believe these errors are all addressed in v1.1.8-7 build

moggieuk, 2023-01-09:
Fixed issue

closed 2023-01-09 by moggieuk