moggieuk/ERCF-Software-V3 #13: Pro tip Installation Issue

NikosSprocket, 2023-01-05:
I tried running the "Pro tip" installation for placing the new configuration files in the /tmp folder but I received the following error:

`Klipper service found
Linking ercf extension to Klipper...
ln: failed to create symbolic link '/tmp/klippy/extras/ercf.py': No such file or directory
`
Which is expected as klipper does not exist under /tmp.

Maybe, introduce a new option like "-t" (for test?) which will clone the klipper existing files under /tmp either with -k and -c custom location or without for the standard location under  `~/printer_data` per new MainsailOS 1.x and link per MainsailOS 0.9x.

moggieuk, 2023-01-09:
Thanks I will address this in future update.  It is a harmless error but it should work nicely.   I'll look into the idea of a "test" install...

NikosSprocket, 2023-01-10:
No worries! Low significance; just an FYI. Thanks!

moggieuk, 2023-01-12:
I've made the fix in my local branch so it will roll up with upcoming v1.1.9 release. Thanks.

moggieuk, 2023-02-02:
Closing because it is fixed.

closed 2023-02-02 by moggieuk