moggieuk/ERCF-Software-V3 #17: Feature: Smart Clog enable

MuddyPuddles4325, 2023-01-18:
I was thinking it would be nice to have the clog detection only enabled when a filament has actually been loaded. I am using the sensorless mode and I find that even after an ERCF home and no filament has been loaded if the clog detection will trigger. It is very minor, I found this when I realized I left the "purge" on during nozzle cleaning before I loaded any filament.

Or even simpler would be to have a macro to enable/disable the clog feature without requiring a klipper restart? Previously I was using a dummy rspi io value to add an enable/disable function to Mainsail.

moggieuk, 2023-02-02:
Good point.   Sometimes you do want clog detection on though, when using bypass for example.  But I take your point and will add -- I have a big update to clog detection in the works now..

MuddyPuddles4325, 2023-03-13:
This is perfect, thank you for all your work

closed 2023-03-13 by MuddyPuddles4325