moggieuk/ERCF-Software-V3 #18: ERCF_TEST_GRIP broken

qratz, 2023-01-18:
Hi there,
ERCF_TEST_GRIP does not work in the current version. This is because TEST_GRIP call SERVO_DOWN and then ERCF_MOTORS_OFF. Unfortunatly the latter calls SERVO_UP, which immediately pulls the servo up again, not allowing the grip to test. A workaround to the test filament grip atm is calling ERCF_SERVO_DOWN and then M84 to disable all steppers.

Nomsplease, 2023-01-19:
Can confirm I am seeing the same issues with this.

in ercf.py I traced this to these lines below:
https://github.com/moggieuk/ERCF-Software-V3/blob/e06db5bc974be0b8815e6162f85bb847c93c5724/extras/ercf.py#L2377-L2382

Where the last line that calls this
`self.cmd_ERCF_MOTORS_OFF(gcmd)`

Triggers the follow block which sets the servo back to ups position.
https://github.com/moggieuk/ERCF-Software-V3/blob/e06db5bc974be0b8815e6162f85bb847c93c5724/extras/ercf.py#L876-L882

Which specifically the issue here is on line 880 where the servo is called back up.
`self._servo_up()`

It looks like turning the motors off is only used for the test grip function, so there should be no harm in removing the line from what I can see. I am new to ERCF so hopefully someone more experienced could chime in here on why we are calling the serve up on this motor off function.

moggieuk, 2023-02-02:
:-)  Late response but this was an omission with a particular point release and was quickly fixed.   It's all good now I believe..

closed 2023-02-02 by moggieuk