moggieuk/ERCF-Software-V3 #21: Unable to parse move 'G1 E--1.4 F2400.0' & Top Hat Lockers not engaged

Durahl, 2023-01-26:
Greetings!

I made the change to the Happy Hare somewhere in the end of 2022 which worked fine but today after making the Update to the most recent version it appears for this having been a classic case or `Never change a running System` 😑

I did both a Manual and then a Scripted Install of the Files

Two problems have come up so far:
- The first one that popped into view would be this Error Message which appears to come up somewhere along the Unloading Process - After the Tip Forming / Before the Unloading:
[image]
The above Image being after I was somehow able to coax it into finishing the print but the Filament still in the Tool Head.

- The second one would be the Servo not behaving properly whenever anything but T0 is active which I kinda suspect to be the root cause for the first Problem...
[image]
[Video of running an ERCF_CHECK_GATES](https://youtu.be/C-LxbQEPx2E)
...as I've observed the Gear Wheel attempting to retrieve the Filament without the Top Hat Locker being engaged 🤔

moggieuk, 2023-02-02:
The servo issue was a PWM timing issue with klipper that was completely eliminated in v121 which contains a custom servo driver --  the pins on the discord channel explain this. 

As for the parsing error it looks like it might be something in your standalone tip forming logic...

Durahl, 2023-02-02:
@moggieuk And where does that negative tip forming logic come from? I checked the files at the time, and none showed any of the values ( as a whole or in parts ) which I interpret as them not coming from the Slicer as otherwise they'd be visible IN the Sliced File which basically means they can only come from somewhere in the revamped Happy Hare Configs ( since I made the initial Install ) - I'd wager a guess somewhere in the Python file hidden as a Formula ( Coding not being my forte )

Another Indicator it being somewhere in there is the fact the System having worked with both the original ERCF Configs as well as one of the earlier Happy Hare ones I installed but failed to make a Backup from before upgrading to 1.2.0... I tried going back as far as 1.5 but then basically had to give up and revert back to ERCF to fix that problem.

moggieuk, 2023-02-04:
In ercf_software.cfg you will find the standalone tip forming macro (this is used when not in a print).  Many of the G1 moves have "-" before the parameter because the actual value of the parameter is expected to be positive.  So make sure all your "variables" at the start of the macro are positive numbers..

moggieuk, 2023-03-24:
I'm going to close out this issue because I believe it was a configuration error.  Please reopen if you still have a problem.

closed 2023-03-24 by moggieuk