moggieuk/ERCF-Software-V3 #22: Need range checking when reading new persisted variables

jwhite, 2023-01-27:
I changed my ercf parameters file to turn on all 6 gates and immediately crashed the printer to restart.
I suspect that there is no range checking on this parameter, so increased it by hand.

Could be an improvement.

ercf_state_tool_to_gate_map = [0, 1, 2, 3]

moggieuk, 2023-02-02:
Yup, probably some issues there because I didn't check changing list sizes.  I'll look into some safety checks. Thanks.

moggieuk, 2023-03-24:
I introduced additional checks on startup so now if there is a mismatch on startup the default values for tool_to_gate_map and other arrays will be used.  The default is either that store in ercf_parameters.cfg OR built in default values.

closed 2023-03-24 by moggieuk