moggieuk/ERCF-Software-V3 #23: ERCF_EJECT fails if filament is in extruder

jwhite, 2023-01-28:
After a print has completed successfully I attempted to change filament by starting with bringing up the extruder temperature and selecting ERCF_EJECT.

Filament was NOT unloaded from the extruder with tipforming.  Observed behavior was:

Servo Down
ERCF Motor in reverse at full speed.

Because the filament was still held by the extruder, this quickly ate away at the filament at the ERCF side and the filament did not move.

Full log is attached:

`$ ercf__eject
17:16:04 
// - DEBUG: Unloading tool T5
17:16:05 
// Unloading filament...
17:16:05 
// ERCF [T5] <<<<< [encoder] <<<<<<<<<<<<< [extruder] ... [sensor] ... [nozzle] (@0.0 mm)
17:16:05 
// - DEBUG: Unloading bowden tube
17:16:05 
// - DEBUG: Setting servo to down angle: 118
17:16:05 
// - - TRACE: Processing idle_timeout Printing event
17:16:05 
// - - TRACE: Determined print status as: complete from print_stats
17:16:05 
// - - TRACE: Determined print status as: complete from print_stats
17:16:05 
// - - TRACE: Servo turned off
17:16:05 
// - - TRACE: Servo turned off
17:16:05 
// - DEBUG: Moving the gear and extruder motors in sync for -10.0mm
17:16:06 
// - - TRACE: Sync unload. Stepper: 'both' moved -10.0mm, encoder measured 10.6mm (delta -0.6mm). Counter: @10.6mm
17:16:17 
// - - TRACE: Course unloading move #1 from bowden. Stepper: 'gear' moved -999.7mm, encoder measured 11.3mm (delta 988.3mm). Counter: @22.0mm
17:16:17 
// ERCF [T5] <<<<< [encoder] <<<<<<<...... [extruder] ... [sensor] ... [nozzle] (@22.0 mm)
17:16:17 
// - - TRACE: Determined print status as: complete from print_stats
17:16:17 
// - - TRACE: Determined print status as: complete from print_stats
17:16:17 
// - DEBUG: Setting servo to up angle: 20
17:16:17 
// - - TRACE: Servo turned off
17:16:18 
!! An issue with the ERCF has been detected whilst out of a print
17:16:18 
// Reason: Failure to unload bowden. Perhaps filament is stuck in extruder. Moved 999.7mm, Encoder delta 988.3mm
17:16:18 
// - - TRACE: Processing idle_timeout Ready event`

ERCF_HOME works as expected, ejecting the filament by first using tipforming and the eject at the extruder.

jwhite, 2023-01-28:
My diagnosis is that I loaded the filament by hand, so the previous state must have been :

self.LOADED_STATUS_PARTIAL_END_OF_BOWDEN or self.LOADED_STATUS_PARTIAL_HOMED_EXTRUDER:

Because of this the tip forming unload was not done.

I think this state occurs with some frequency when a load fails, manual intervention then allows the extruder to work, but the state is left in the PARTIAL_END_OF_BOWDEN condition.

A possible fix might be to do the tool forming eject at any time the sensor is active, or an ERCF_EJECT has occurred to fix a load.

jwhite, 2023-01-28:
One reason this is particularly severe is that it is guaranteed to chew a half moon shaped hole in the filament.... about 1 meter UP from the extruder. And then it is hidden during the real unload.  

It's like a ticking time bomb for a failed print when that meter finally makes it to the extruder.... probably in a future print!

moggieuk, 2023-02-02:
If I'm reading your log correctly the starting state is incorrect (because you loaded into the extruder by hand?). I think you said that but wanted to be sure.

If that is the case, then yes, the next point at which Happy Hare would detect a problem is when the fast bowden move fails.
The state really needs to be reset with:
ERCF_RECOVER command.  Either in automated way or force it with ERCF_RECOVER LOADED=1

Since you have a toolhead sensor there could be an additional check before the bowden move which would be a good addition...

moggieuk, 2023-02-05:
This is fixed now, so what out for the v1.2.2 release.

closed 2023-03-07 by moggieuk