moggieuk/ERCF-Software-V3 #27: ERCF_CHECK_GATES crashes klipper when selector blockage is detected

qratz, 2023-02-08:
I am in the process of tuning sensorless homing on the selector. While doing so i realized that ERCF_CHECK_GATES crashes klipper when selector blockage is being detected:

`Internal error on command:"ERCF__CHECK_GATES"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 425, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 186, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/ercf.py", line 2930, in cmd_ERCF_CHECK_GATES
    self._select_gate(gate)
  File "/home/pi/klipper/klippy/extras/ercf.py", line 2275, in _select_gate
    self._move_selector_sensorless(offset)
  File "/home/pi/klipper/klippy/extras/ercf.py", line 2199, in _move_selector_sensorless
    raise ErcfError("Selector path is probably externally blocked")
ErcfError: Selector path is probably externally blocked`

moggieuk, 2023-03-07:
I believe this has been addressed with version v1.2.3 
Can you confirm?

moggieuk, 2023-03-24:
I'm sure this is fixed, so closing issue.

closed 2023-03-24 by moggieuk