moggieuk/ERCF-Software-V3 #29: Filament switch before extruder gears?

drelich, 2023-02-11:
Hi,

Would like to use my existing filament switch I made for CW2 that is located above the extruder gears. Filament trips the switch, extruder gears start spinning, feeder pushes the filament a set amount of distance so it gets past the gears and then disengages. Kinda like the Prusa's MMU2S with MK3S works. 

At the moment, the calibration fails with 'filament detected before it entered extruder'. I was unable to make the collision or stallguard homing method work (it is marked as experimental so not sure if it's even supposed to be working).

Thanks.

David.

moggieuk, 2023-03-24:
This isn't possible as designed.  With this setup, I would recommend turning off the filament sensor and defaulting to sensorless operation.

A sensor before the extruder simply changes too much in the logic for loading and isn't any doesn't offer any value.  E.g. you cannot conclude that filament is in the extruder if this sensor is active, only that filament is in the bowden. Similarly it offers a homing point, but one that is unlikely to be accurate since the transition from extruder gear into extruder is the most variable step.

Therefore I'm going to close this as 'not planned' because sensorless will provide the needed functionality.  Also, so long as the filament sensor isn't called "toolhead_sensor" it could remain separate from ERCF and allow the user to leverage in macro wrappers for additional functionality / checked.

moggieuk, 2023-03-24:
Not planned

closed 2023-03-24 by moggieuk

drelich, 2023-03-24:
Thanks for taking the time to respond! Your argument seems good enough for me :) Found a CW2 component with a mechanical switch bellow the gears and am happy with it. Cheers.