moggieuk/ERCF-Software-V3 #35: Enable a variable number of attempts to load sensorless toolhead?

mrjoeyjiffy, 2023-02-27:
My most common need for intervention is a failed to load toolhead, almost 100% of the time if I do an eject and reload that same gate it works the second time. Is there a way to make happy hare attempt this a certain number of times before requiring intervention?

moggieuk, 2023-03-24:
Ok, this is possible but will require some careful thought.  There are already lots of retries, albeit not any that do a complete eject first.

moggieuk, 2023-04-03:
I've been looking into this and have experimented and there are two many corner cases and cases where bad things could happen.  I think the best solution is a user implemented retry.  This would be relatively simple to wrap the ERCF_CHANGE_TOOL macro with one that calls it. Checks the "printer.ercf.is_locked" to see if there was an issue and then calls ERCF_CHANGE_TOOL again...

moggieuk, 2023-06-06:
Going to close this issue because the need will become mute with update and completely synchronized gear and extruder steppers!

closed 2023-06-06 by moggieuk