moggieuk/ERCF-Software-V3 #39: Suppress filament actions while printer is printing (or endless spool is in affect)

nigelpjames, 2023-03-26:
I randomly hit eject (macro button) during a print and it unexpectedly ejected the filament. I assumed filament actions would be rejected if the printer wasnt in a pause, error state or had just been unlocked and wasnt expecting it to interrupt the print to do so.
It was stupid and might be an edge case not encountered or tested....it might also be operating as intended ;-)

HH v1.2.3-11-g4d3383f 
Klipper v0.11.0-86-g6026a99a

moggieuk, 2023-04-04:
Hi Nigel, thanks for the note.  Yes it could be more robust.  There are some things that are ok to do when printing (TTG map, changing config, etc) but the one that break the print like eject, gate change could be prevented...

moggieuk, 2023-04-08:
I'm going to close this as "not planned" because although good in concept, it's not possible to implement.

closed 2023-04-08 by moggieuk