moggieuk/ERCF-Software-V3 #40: Cannot Home Selector Using Senseless Homing

sabaatworld, 2023-03-31:
I've installed Happy Hare on my new ERCF and trying to get sensorless selector homing to work. For some reason, it always fails with the following output:

```
2:19 AM
An issue with the ERCF has been detected whilst out of a print
Reason: Homing selector failed because error. Klipper reports: Homing selector failed because of blockage
2:19 AM
ERCF [T?] <.... [encoder] ............. [extruder] ... [sensor] ... [nozzle] UNLOADED (@0.7 mm)
2:19 AM
ERCF [T?] <.... [encoder] ............. [extruder] ... [sensor] ... [nozzle] UNLOADED (@0.0 mm)
2:19 AM
Unknown filament position, recovering state...
2:19 AM
Homing ERCF...
2:19 AM
You are running an old calibration version.
It is strongly recommended that you rerun 'ERCF_CALIBRATE_SINGLE TOOL=0' to generate updated calibration values
2:19 AM
ERCF_HOME
```

Attaching configuration files
[ercf_parameters.cfg.txt](https://github.com/moggieuk/ERCF-Software-V3/files/11120323/ercf_parameters.cfg.txt)
[ercf_hardware.cfg.txt](https://github.com/moggieuk/ERCF-Software-V3/files/11120324/ercf_hardware.cfg.txt)
[ercf_software.cfg.txt](https://github.com/moggieuk/ERCF-Software-V3/files/11120326/ercf_software.cfg.txt)

sabaatworld, 2023-04-01:
I figured out the issue. It was being caused cause the steppers were configured for reverse direction. After removing `!` from `dir_pin`, sensorless homing worked. However, it seems to be always auto-correcting after selecting a tool. For example, it would auto-correct after homing sequence. I say auto-correct because of the following logs:

```
05:46:59 Homing ERCF...
05:46:59 Unknown filament position, recovering state...
05:46:59 - - TRACE: (Toolhead sensor does not detect filament)
05:46:59 - DEBUG: Checking for filament in encoder...
05:46:59 - - TRACE: (Toolhead sensor does not detect filament)
05:46:59 - DEBUG: Setting servo to down angle: 120
05:46:59 - - TRACE: Determined print status as: standby from print_stats
05:47:01 - - TRACE: After buzzing gear motor, encoder moved 0.00
05:47:01 - DEBUG: Filament not detected in encoder after buzzing gear motor
05:47:01 ERCF [T?] <.... [encoder] ............. [extruder] ... [sensor] ... [nozzle] UNLOADED (@0.0 mm)
05:47:01 - DEBUG: Filament already ejected
05:47:01 - DEBUG: Setting servo to up angle: 30
05:47:01 - - TRACE: Setting ERCF gear motor step ratio to 1.000000
05:47:01 - DEBUG: Moving up to 188.0mm to home a 9 channel ERCF
05:47:04 - DEBUG: Check #1 of gear_endstop: True
05:47:04 - DEBUG: Selecting tool T0 on gate #0...
05:47:06 - - TRACE: Selector moved 4.1mm of intended travel from: 0.0mm to: 4.6mm (delta: 0.5mm)
05:47:06 - - TRACE: Truing selector 0.5mm to 4.6mm
05:47:07 - - TRACE: Setting ERCF gear motor step ratio to 1.000000
05:47:07 Tool T0 enabled
```

Where my parameters look like:
```
colorselector: 4.6, 24.8, 45.6, 72.0, 92.8, 114.4, 140.8, 161.6, 182.4
```

sabaatworld, 2023-04-01:
https://user-images.githubusercontent.com/2452471/229258300-8b20d3cb-e029-41b2-9ec1-45a822a9a1e6.MOV

moggieuk, 2023-04-02:
Glad you found the issue.   What you described can also be a result of what I said, but in this case it was a simple fix!

I'm going to close out this issue now. Good luck.

closed 2023-04-02 by moggieuk