moggieuk/ERCF-Software-V3 #50: Toolhead sensor before extruder

iguanahotsauce, 2023-05-05:
I'm running my ERCF on my Prusa MK3S+ Bear that is using a SKR Mini e3 V3 and Klipper, I've been running into an issue where I keep getting the error "Toolhead sensor malfunction - filament detected before it entered extruder!." when trying to load filament and I realized it is because my toolhead sensor is right before the extruder gears and not after the extruder gears. I took a look through the documentation and I couldn't find anything about switching the software so it knows the toolhead sensor is before the extruder, is this something that I am able to do with the ERCF-Software-V3?

chujian, 2023-05-16:
I have the same problem，when I use orbiterV2 Sensor, the sensor is on top of the extruder

moggieuk, 2023-06-08:
ERCF needs to understand the physical capabilities of your printer and currently does not understand or leverage a sensor before the extruder gears.  It is possible that in the future a new loading strategy that could leverage such a sensor could be used (i.e. home filament to sensor before extruder then synchronously load).  The problem is one of accuracy of the loaded filament length.  You need an accurate home position and a position outside of the extruder may not be accurate enough.

At the moment the filament detection logic or the sensor itself should be disabled when using ERCF -- a simple click in Mainsail or Fluidd or you can add to your MMU macros.

I'm going to leave this "issue" here to track developments.

moggieuk, 2023-06-15:
This may be a future option so leaving this "issue" open.

smwoodward, 2023-07-02:
I can add that I would also want with my ratrig.

moggieuk, 2023-07-26:
Ok.  This has been implemented in the upcoming Happy Hare v2 project (it isn't a natively supported switch but the provided customization makes it quite easy to add).   Therefore I'm going to close out this bug as "WILL NOT FIX" because it IS IMPLEMENTED in the upcoming overhaul of Happy Hare.

closed 2023-07-26 by moggieuk