moggieuk/ERCF-Software-V3 #53: Purge Bucket

lednicazar, 2023-05-09:
I'm having some issues to use the purge bucket in the color changes.
In SuperSlicer I've set under the GCODE for Tool Change:
"
T[next_extruder]
CLEAN_NOZZLE
"
Seems to do the process but when the part is taller, the nozzle crash to the part like is loosing the Z. Anyone is using the purge bucket + ERCF?

lednicazar, 2023-05-09:
Temporaly solved:
I made an additional macro, and call that macro instead of T{next_extruder} from SuperSlicer:
"
[gcode_macro ERCF_CHANGE_MYVORON]
gcode:
    #G1 E-1.0 F300                 ; retract filament
    {% set TOOL = params.TOOL|default(1)|int %}
    {% set STANDALONE = params.STANDALONE|default(1)|int %}
    # safe anti-stringing move coords
    {% set th = printer.toolhead %}
    {% set x_safe = th.position.x + 20 * (1 if th.axis_maximum.x - th.position.x > 20 else -1) %}
    {% set y_safe = th.position.y + 20 * (1 if th.axis_maximum.y - th.position.y > 20 else -1) %}
    {% set z_safe = [th.position.z + 20, th.axis_maximum.z]|min %}
    G0 X{x_safe} Y{y_safe} Z{z_safe} F20000  ; move nozzle to remove stringing

    ERCF_CHANGE_TOOL TOOL={TOOL} STANDALONE={STANDALONE}
    
    CLEAN_NOZZLE
    #Safe_possition
    G0 X{x_safe} Y{y_safe} Z{z_safe} F20000  ; move nozzle to remove stringing
"
With that, I move first to a safe possition, then change the filament, clean the nozzle, return to a safe possition and resume.

moggieuk, 2023-06-05:
I think this is the correct approach.  If you intercept the tool_change procedure then you have to take responsibility for any movements that occur.  Note that the default Happy Hare "pause" functionality will lift the z by 5mm before calling the PAUSE macro.  This is for a very similar reason.

moggieuk, 2023-06-06:
Closing as works as intended

closed 2023-06-06 by moggieuk