moggieuk/ERCF-Software-V3 #60: [FR] Possibility of replaying lost moves due to clog?

thisiscam, 2023-05-24:
When I ERCF into a clog, I can manually declog and resume the print. But the last few mms of extrusion will be lost due to clogging. 
I would love the possibility to analyze the running gcode file and trace back a few lines and restart at where the clog might have happened.

moggieuk, 2023-06-05:
That is certainly advanced!   It would be difficult to know how far to replay.  There is a delay between clog and when it is detected.  The clog detection length (which is kept as short as possible with the auto tuning) is the trigger point.  In theory this gives an approximation of how far to go back in history.  Arguably this would be better than a "gap" in the print but it would equally as well lead to "double printing"... hmmm

thisiscam, 2023-06-05:
> That is certainly advanced! It would be difficult to know how far to replay. There is a delay between clog and when it is detected. The clog detection length (which is kept as short as possible with [...]

Yup might be difficult and might need to somehow hook into klipper. But usually double printing will be ok --- it's not as good as a successful print, But likely reduces chance of spaghetti and missing lines due to lost moves

Surion79, 2023-06-15:
Clogs must not only be massive 'blocks' at a exact point of time. It could also result in bad print quality before clog has been triggered.
And clog is not always clog. Could also be that the spool/filament has some major friction. So a clog is not necessarily a real clog.
So identifying what has happened and how to or whether replay is, will be a challenge

thisiscam, 2023-06-21:
> Clogs must not only be massive 'blocks' at a exact point of time. It could also result in bad print quality before clog has been triggered.
> And clog is not always clog. Could also be that the spool/filament has some major friction. So a clog is not necessarily a real clog.
> So identifying what has happened and how to or whether to replay will be a challenge.

Maybe could let the user decide whether to roll back, or how much to roll back? AFAIK, the user must intervene after a clog anyway, and one could have replayed the lost gcode manually. I'd propose to make that process less tedious.

Surion79, 2023-06-21:
I agree, that a user must be able to decide what to do after a clog pause regarding replayability.

moggieuk, 2023-09-04:
It feels like this is beyond the current scope of the Happy Hare project.   It is certainly something to experiment with, but ideally it would be visual .. perhaps the gcode visualization in Mainsail... "rewind to this point and restart" kind of functionality.