moggieuk/ERCF-Software-V3 #65: Disable PA during loading/unloading

thisiscam, 2023-06-09:
I was looking at [klipper PA docs](https://www.klipper3d.org/Kinematics.html#extruder-kinematics), and my understanding is that PA changes the speed of any single extruding and retraction. 
This makes me wonder if PA will actually cause less/more moved distance of the filament?

If PA indeed introduces a small "error" to the moved distance, the consequence of enabling PA during loading and unloading then would be that we get a small inaccuracy of the filament movement. 

These said,
- I am not 100% positive that PA introduces an error.
- I am not sure if this amount of inaccuracy is basically ignorable.

moggieuk, 2023-06-15:
I'm ignorant... does PA cause a change in the overall volume of filament pushed or just alter the flow rate during acceleration and deceleration?   If the latter then this difference will have no net effect on syncing..?

thisiscam, 2023-06-15:
This is the plot of PA taken from the klipper docs above:

[image]

I agree that the overall volume of filament being pushed and pulled stays the same if there there is no filament change. 
This is because the pushing and pulling always cancels out as long as a print is going on.

But, if the filament goes in and out of the extruder, my understanding is that there will be a small delta. 
Consider if the filament is above the extruder, and we start doing a synced load with PA enabled. And for simplicity, consider the old syncing behavior where the gear does not have PA but the extruder does.
My understanding is: 
- The extruder will do a "speedup PA move" (the leftmost green concave in the plot) when the filament has not entered the extruder. 
- Once the filament gets in and the extruder comes to a stop, the extruder will take a "braking PA move" (the rightmost green convex in the plot), thereby causing the filament to travel less distance than it is supposed to.
- Because the initial speedup PA move wasn't acted on the filament, we ended up extruder a tad bit less than we want.

Does that make sense?

With our new syncing logic, this error might be alleviated somewhat, but the behavior is still not ideal, since we are pushing a noodle with the gear and not much we can do about precision there. 

With a toolhead sensor, this might not be a problem since we are going to home to the sensor first (and the inaccuracy is consumed at that point)

Note: this analysis is done purely by looking at the above plot from klipper docs. I have no idea if that plot is an accurate description of actual implemented PA, or if the delta is significant enough for us to care.

Surion79, 2023-06-28:
if you do single color print with ercf with this in mind, this could mean that the tension between the toolhead and the sync loader can accumulate? There is no filament swap that can negate such a buildup.

thisiscam, 2023-06-29:
> if you do single color print with ercf with this in mind, this could mean that the tension between the toolhead and the sync loader can accumulate? There is no filament swap that can negate such a buildup.

If you mean if the not disabling PA could cause any issue during a single color print --- I don't think so. The errors are canceled out for each extrude and retract movement.

moggieuk, 2023-07-26:
It would certainly be easy to add a "record and set PA to 0" at start of load and restore at end but I'm thinking we are talking sub mm effect here.   It can also be done in gcode by wrapping the `CHANGE_TOOL` macro.

If we can demonstrate this is an issue I can formally add the logic in Happy Hare (v2 now of course)