moggieuk/ERCF-Software-V3 #67: Syncronized Gear Stepper with TMC2209 gear stepper and TMC2240 toolhead valid?

Surion79, 2023-06-10:
I am currently facing the issue that I can't use ERCF with the error that the gear_stepper can't be contacted. I change the TMCs and it is the same error. 

`Unable to read tmc uart 'gear_stepper' register IFCNT`

I am not sure if it is related to my setup with different tmc types and connection UART/SPI.

thisiscam, 2023-06-12:
AFAIK, the syncing feature should work. Maybe you could try it and report back? Try to enable the sync feature, and without loading any filament do: `ERCF_SYNC_GEAR_MOTOR sync=1 servo=1`. Then extrude under your web UI, see if the gear moves at all?

 The message `TMC driver not found for extruder, cannot use current increase for tip forming move` is a different error on changing the current during tip forming moves. It seems like ERCF doesn't support tmc2240 for some reason: https://github.com/moggieuk/ERCF-Software-V3/blob/6afa9929800aa645f5958d7f0240d17b0c94b86e/extras/ercf.py#L539 . @moggieuk any ideas?

moggieuk, 2023-06-15:
I may have just left tmc2240 out of the list (I don't have any controllers other than 2209's to test so I did this from the Klipper doc)..  I will add.  Note the error message is only for the ability to reduce current on tooltip move, it will not effect the synchronized gear/extruder movement.

One slight concern. Klipper doc says:
SET_TMC_CURRENT[¶](https://www.klipper3d.org/G-Codes.html?h=tmc#set_tmc_current)
SET_TMC_CURRENT STEPPER=<name> CURRENT=<amps> HOLDCURRENT=<amps>: This will adjust the run and hold currents of the TMC driver. HOLDCURRENT is not applicable to tmc2660 drivers. When used on a driver which has the globalscaler field (tmc5160 and tmc2240), if StealthChop2 is used, the stepper must be held at standstill for >130ms so that the driver executes the AT#1 calibration.

I don't ensure a delay of >130ms before changing current... I cannot test this...

moggieuk, 2023-06-15:
Ok. tmc2240 support is pushed.  Caveat about possible issue highlighted in previous comment.  Open a new issue if you have problems.

closed 2023-06-15 by moggieuk

Surion79, 2023-06-15:
I am currently waiting for a replacement board. then i can test this and report back to you :)