moggieuk/ERCF-Software-V3 #72: Stallguard homing position?

thisiscam, 2023-06-14:
Opening this as a reminder: 
I think the gears stepper's position should be set to 0 (or some known value) before [this homing move](https://github.com/moggieuk/ERCF-Software-V3/blob/6afa9929800aa645f5958d7f0240d17b0c94b86e/extras/ercf.py#L1823)?

Based on [this discussion](https://klipper.discourse.group/t/trying-to-understand-homing-state-home-rails-parameters/5460), `movepos` is the target position at the homed location, and the stepper basically moves from its current position towards `movepos` until it hits the end stop, then it will force set its position to `movepos`.

Just ran across this without any concrete evidence that this might be an actual bug, since I don't use stallguard homing right now.

moggieuk, 2023-06-16:
Thanks.  I think this experimental homing option is only used by one person.  It really isn't much of an advantage / more reliable to just looking at the encoder for movement to deduce homing to extruder.  The more sophisticated way is to use a toolhead sensor.  It requires additional stallguard setup and also is problematic with EASY-BRD.

I haven't tested for a LONG time.

moggieuk, 2023-07-26:
In light of the fact that syncing / homing development is moving to the Happy Hare v2 release I'm going to close out this bug hear because it already has been addressed in HHv2.

closed 2023-07-26 by moggieuk