moggieuk/ERCF-Software-V3 #78: Sensorless homing (selector) and travel delta when moving.

cmroche, 2023-06-26:
I'm re-configuring my ERCF with Happy Hare and in the initial steps of tuning the Stallguard threshold. I've found the least sensitive value that get my selector able to move through the entire range of carts, but have had some issues with false-positives for something blocking "inside" due to a larger then expected movement delta.

Typically I'm getting 1.2mm - 1.3mm delta, rarely as much as a 2.0mm delta. Once I relaxed the delta check everything "appears" to be fine, but I'd like to understand better what's actually causing this.

@moggieuk Do you have any insight?

```
08:35:24 - DEBUG: Selecting tool T3 on gate #3...
08:35:24 - - TRACE: Determined print status as: standby from print_stats
08:35:25 - - TRACE: Selector moved 24.3mm of intended travel from: 45.5mm to: 72.5mm (delta: 2.7mm)
08:35:25 - - TRACE: Truing selector 2.7mm to 72.5mm
08:35:25 - - TRACE: Setting ERCF gear motor step ratio to 1.000000
```

cmroche, 2023-06-26:
I was able to track this down.

Basically it's caused by the fact that `homing_move` is used (which makes sense), Stallguard doesn't work very reliably near 0 velocities and basically what happens when the selector decelerates to stop at the target position, if it goes slow enough due to a low acceleration (I was using 600), it causes Stallguard to trigger a bit early.

With 600 m/s^2 I was getting the 1.2ms - 1.4ms I mentioned above.
With 1500 m/s^2 I am getting around 0.5ms each time.

Since this can vary a lot based on configuration, perhaps the default 1mm tolerance should be changed to something a bit larger? When using sensorless it might be good to advocate for a higher default acceleration on the selector too.

moggieuk, 2023-07-26:
Wow. Don't know how I missed this message before.  Very interesting detective work.  I was never able to find the correlation between different deltas.  Well done 👏 . 

I will experiment with increasing the default acceleration. 1500 seems a little high but can ship with higher and definitely document this finding.

BTW I have a major rework of Happy Hare in a new repo.  I'll accommodate this feedback there..

moggieuk, 2023-08-27:
Ok. For Happy Hare v2 (which will replace this software) I have made default acceleration 1200mm/s^2 and increased tolerance to 1.5mm.  I this brings everything back into the happy zone.

I'm going to close this issue on this release (because I'm unlikely to update given the HHv2 replacement).

closed 2023-08-27 by moggieuk