moggieuk/ERCF-Software-V3 #81: Multiple Klipper Installations on One Pi

AustinTylerDean, 2023-07-06:
Hi there,

Sorry for the Sesame Street post, but I've searched around the best I could for an answer, AND spent some quality time trying to figure it out, alas the install.sh isn't working for me.

I have two instances of klipper/moonraker installed via KIAUH. Only one is active at the moment. I believe the service I have active is not "klipper.service" but "klipper-1.service".

Although this isn't very common, I can see some people having trouble with this in the future if they want to use the install script. I'm used to manually setting up all my config files, but the "gear stepper stealing the selector end switch" option just blew my brains out, so I'm going a conventional route.

Outside of someone fixing this via a PR, could anyone guide me through as to how many places I need to change this in the install.sh?

Thanks

AustinTylerDean, 2023-07-08:
> Isn't it enough to run the install script with the -c and/or -k parameter
> as described on the repo readme?

No, as your script looks for a klipper service that does not exist with a
multiple printer installation. Change it to klipper*.service and I think
that will work.

AustinTylerDean, 2023-07-16:
I would love to contribute but lack the confidence and skills to do so. The best I can do is summarize which details that need to be changed.

In install.sh change the following:
`KLIPPER_CONFIG_HOME="${HOME}/printer_data/config"
`
needs to accommodate "printer_n_data" where n = an integer that Kiauh uses to designate multiple printer installations.

also

`if [ "$(sudo systemctl list-units --full -all -t service --no-legend | grep -F "klipper.service")" ]; then
`
needs to change to "klipper*.service" or equivalent to accommodate multiple printer installations as the services are numbered with the corresponding integer as mentioned previously.

I also lack the skills to understand why my installation procedure allowed the Happy Hare installation to assign the ownership of the directory to root instead of the user (perhaps I used sudo when I didn't need to?) If anyone struggles with the update_manager coming back as invalid, see my associated closed issue.

AustinTylerDean, 2023-07-21:
If anyone is struggling with multiple Klipper installations, all you have to do is go to ~/ERCF-Software-V3 and edit install.sh with the above changes I have mentioned. You can also fork the installation on Github, make those changes, and then make sure to swap your name with Moggie's in the url under update_manager and you won't have to deal with a "dirty" installation. I'm pretty sure changing that install.sh will do it.

Good Luck!

moggieuk, 2023-07-26:
If you can submit a PR for the suggested fix I can take a look.   You are welcome to fork but then will run into problems getting update / bug fixes..

moggieuk, 2023-07-26:
Question: what is the moonraker service names when you have multiple?

Is this just klipper and moonraker?

AustinTylerDean, 2023-07-26:
> Question: what is the moonraker service names when you have multiple?
> 
> Is this just klipper and moonraker?
> 
> Perhaps you can run these commands and attach results: systemctl list-units --type=service | grep klipper
> 
> systemctl list-units --type=service | grep moonraker

[image]

moggieuk, 2023-09-04:
Thanks for the additional info.  I will take a look when I get a chance but I will add to Happy Hare v2 rather than this release...