moggieuk/ERCF-Software-V3 #82: Moonraker.conf Creates INVALID

AustinTylerDean, 2023-07-09:
Hi There! I'm trying to figure out why my installation is not finding the git version, which is what I assume the problem is.

[image]

[image]

I've copied and pasted the update_manager portion, and I have 6 other managers working fine, so connection doesn't seem to be the issue. Is there something specific you have to do in github to allow calls for the current version number?

My home directory is correct...
[image]

Any ideas?

AustinTylerDean, 2023-07-11:
[image]

Like I said, I thought my home directory was okay...

Surion79, 2023-07-12:
`pi` please. ;)

AustinTylerDean, 2023-07-13:
What does my username have to do with it?

kolbenhans, 2023-07-14:
your klipper, moonraker etc are also stored under this user & home directory, in that case it should working correct i guess.

AustinTylerDean, 2023-07-16:
The answer is that ~/ERCF-Software-V3 was owned by root, not the user. Solution was to putty into ~ and run "sudo chown -R <username> ~/ERCF-Software-V3"

The -R is important, as without it you end up owning a directory but none of the contents... Good ol' Linux.

closed 2023-07-16 by AustinTylerDean