moggieuk/ERCF-Software-V3 #84: Fatal error when changing extruder

ssackett, 2023-07-16:
I have a 12 gate ERCF.  My Voron 2.4 has a single extruder and my GCode uses gates 1, 3, 5, 10 and 11.  The GCode executes through the first layer but when it tries to start the second layer it experiences a fatal error that issues the message 
"The value '11' is not valid for EXTRUDER".  This is despite the fact that it had just finished using gate 11 which is mapped to extruder 11.    This error first started happening after I cleared a non-fatal error, unlocked the ERCF and executed an ERCF_HOME.  This error now occurs whenever I use the printer despite having done a cold boot and re-slicing the project.  Klipper.log is attached.

[klippy (24).log](https://github.com/moggieuk/ERCF-Software-V3/files/12065176/klippy.24.log)

moggieuk, 2023-07-26:
Hi, this error is not part of Happy Hare.  I believe it comes from the gcode parser in klipper.  I suspect there is something in your gcode that is causing this.  I would examine that and then type that command directly on the console and see what happens. 

If you supplied the ercf.log I can tell you what Happy Hare is seeing...

ssackett, 2023-07-26:
Hi,

I set the persistence level to zero and it hasn’t happened again.  I will send you the log if it happens again.

Thanks,

Steve

moggieuk, 2023-08-27:
Thanks.  I can't see why persistence_level in Happy Hare would make any difference because as I noted I confirmed that this error is generated by gcode.py in klipper.   If you parsed the gcode file looking for "EXTRUDER11" or similar.

I'm going to close this issue since I'm 99.9% sure it is not Happy Hare, but please reopen if you learn more..

closed 2023-08-27 by moggieuk

ssackett, 2023-08-27:
Moggieuk,

Thanks for all the work you have done writing Happy Hare.  I know how hard it is to write and maintain software and documentation and I appreciate your efforts.

I have a Voron 2.4 and 12 gate ERCF both of which are highly modded mechanically but not in ways that effect the software.  I also have custom made filament buffers that I use instead of ERCP (see links).  These are made from laser cut acrylic sheets and some 3D printed parts.

I am not asking for help with this next part except for one request, I just want to tell you what I am working on and perhaps send you information about my solution since you may run across others having similar problems.  My one request is that you provide a software option to synchronize the ERCF and LGX motors during tip forming, I will explain why below.

Currently I am trying to get the ERCF to work with a Bondtech LGX extruder and ACE hot end combination.  After several failed attempts, based on adjusting the tip forming and skinnydip parameters, I have come to the conclusion that I need an additional software feature which is to synchronize the ERCF and LGX during tip formation. This may be helpful because I think the LGX does not have enough power to drive some of the moves required to do tip forming.

When I do standalone tip forming I am able to get good tips but when I hook up to the ERCF and try to print the ACE clogs after a few tool changes.  These clogs form at the upper throat of the ACE cooling tube where it transitions to the plastic body of the LGX.  To me this means that the LGX is failing to move the filament in the same manner when connected to the ERCF as it does when doing standalone tip forming.  The most obvious conclusion is that the LGX is not able to handle the extra load placed on it by driving the filament backward into the ERCF (of course I could be wrong about this).  One thing that occurred to me is that this problem could be fixed by synchronizing the ERCF with the LGX during tip forming to lighten the load.  If Happy Hare already has this feature please point it out to me in the docs.

Just to elaborate, I think the LGX lacks the torque needed to do the rapid moves required for breaking the filament and skinny dipping and this results in pulling molten filament into the upper throat of the ACE cooling tube where it solidifies against the cool plastic surface of the LGX.  There is probably a bit more to it than this because I have used the same LGX on a different toolhead that used an E3D V6.  

My guess is that the short cooling tube of the ACE doesn’t do as good a job cooling the filament as the V6 does, making it more sensitive to load conditions.  The V6 cooling tube is so long that even if the filament was still molten that it would probably solidify during the time it takes to extract it, while simultaneously shaping the tip adequately.  Just guessing but it seems plausible. 

I am working on mechanical mods to the ACE to improve the cooling in case that helps.  The cooling tube isn’t bonded to the heatsink on the ACE and I believe the cooling can be improved by squeezing the heatsink against the cooling tube.  In order to accomplish this I am going to try slitting the heat sink and drawing it tight against the cooling tube with screws.  Also I am going to try wrapping the cooling tube with copper wire and connecting that to the heatsink.  You never know it might work.

Steve

PS.  I modded the ERCF several ways listed below but one interesting mod is I developed a filament gate that doesn’t use magnets.  I had trouble getting the magnetic washer gates and return springs to work so I redesigned the gates so they aren’t required.  A benefit of this is that the ERCF can be mounted in any orientation.  Let me know if you want the CAD drawings for this.

1. Added a stiffner to hold the gates in line.
2. Added a larger servo with a spring loaded plunger. 
3. Larger filament drive motor.
4. Linear ball bearings rather than slip bearings.
5. Metal drive gear rather than plastic.
6. Magnetic shaft encoder.
7. Tensioning spring to apply pressure to encoder drive. 

——————————

Filament buffers: https://youtu.be/F6AMwDrCMpk?si=vA48J16zsPx6_RZk
https://youtu.be/P7j3j8DdvVo?si=07d6VK0iYK2q_poS

LGX/ACE: https://www.bondtech.se/product/lgx-ace-mosquito-printhead/?gclid=Cj0KCQjw6KunBhDxARIsAKFUGs8S8ul0RtkSkYGXomva6VAbaWif4lmoT5GsOT5T4URzLhp-JIwb72oaAjU-EALw_wcB