moggieuk/ERCF-Software-V3 #87: Add compatibility with Spoolman

lettore, 2023-08-03:
Hello, I found an interesting project that can be used as Spool Manager, Spoolman. This is useful for managing and tracking filament usage, Moonraker is fully supporting it. Using Spoolman as spool data storage backend will be awesome and really easy, just set the spool ID number that is inserted in the gate, ERCF can then take filament info like material and color from Spoolman. When loading a specific tool ERCF should say to Moonraker which filament ID is using and Moonraker itself will update the filament usage on Spoolman.

lettore, 2023-08-18:
For now I'm using this method, I set the material type as the ID number of the spool instead of setting PLA etc.
Then in ercf_software.cfg
```
[gcode_macro _ERCF_ACTION_CHANGED]
description: Called when an action has changed.
gcode:
    # This occurs when the ERCF action status changes.  `printer.ercf.action` will contain
    # the current action string. See Happy Hare README for full list
    #
    # This could be a place to set LED status or similar. The logic here
    # demonstrates the three major action states
    {% set ACTION = printer.ercf.action|string %}

    {% if ACTION|string == "Idle" %}
        # Add your logic here
        # _STATUS_STANDBY
        {% if printer.ercf.tool|int > -1 %}
            #if material loaded is a number, then it should be a Spoolman ID
            {% if printer.ercf.material|int > 0 %} 
                #if material loaded is a number, then it should be a Spoolman ID
                {% if printer.ercf.filament == "Loaded" %} 
                {% set id = printer.ercf.material|int %}
                #set the spool ID
                {action_call_remote_method("spoolman_set_active_spool",spool_id=id)}
                {% else %}
                #clear spool ID as filament is not loaded
                {action_call_remote_method("spoolman_set_active_spool",spool_id=None)}
                {% endif %}
            #clear spool ID as it's invalid
            {% else %}
                {action_call_remote_method("spoolman_set_active_spool",spool_id=None)}
            {% endif %}     
        {% endif %}     
    {% endif %}
    
    {% if ACTION|string == "Loading" %}
        # Add your logic here
        # _STATUS_LOADING
    {% endif %}

    {% if ACTION|string == "Unloading" %}
        # Add your logic here
        # _STATUS_UNLOADING
        {action_call_remote_method("spoolman_set_active_spool",spool_id=None)}
    {% endif %}
```
For now it's working quite well

moggieuk, 2023-09-04:
Thanks.  This is on the intended feature list for Happy Hare v2.  Soon....