moggieuk/ERCF-Software-V3 #88: Feature Request(?): Is it possible to have two klipper instances

SHKinsem, 2023-08-08:
Had a little discussion with my friends and found that the single-thread feature of Klipper is really annoying. Then, suddenly it occurred to me: why don't we use another Klipper instance as the container for the ercf software? In that way, with two instances, we would have two threads that can operate asynchronously. For example, in the endless spool mode, we wouldn't have to wait for the feeder to eject and reload filament. Instead, it could move the selector directly to the desired filament box and feed it without any pause.

So, there is only one problem as far as I am concerned: is it possible to pass variables between multiple Klipper instances?

Or, anyway, is there a downside to doing this?

moggieuk, 2023-09-04:
While an interesting concept synchronizing two processes is incredibly difficult.  Think of all the combinations of stepper --> mcu --> klipper combinations!   Synchronized gear movement would not be possible, etc.

SHKinsem, 2023-09-05:
> While an interesting concept synchronizing two processes is incredibly difficult. Think of all the combinations of stepper --> mcu --> klipper combinations! Synchronized gear movement would not be possible, etc.

But, on the other hand, if we take bambulab as a reference, they do not synchronize both extruders by using a limit switch and a filament buffer. The limit switch triggers when the buffer runs out of filament, at which point the MMU (Multi-Material Unit) will extrude more to fill the buffer. I believe that this approach offers even greater reliability than directly synchronizing two extruders, which might have different gear ratios.