moggieuk/ERCF-Software-V3 #90: Fwature Request: Retry on load failure

vertigo235, 2023-08-10:
I'd like to recommend that when a load failure occurs, like a clog detected, or too much slippage, etc....  That you allow the option for the ERCF to retry up to a defined amount of retries.

I suggest this because after many years of using the MMU, and ERCF I have found that often at least 25-30% of the time, this just works.

A lot of times it goes like this, printer paused and requests attention on say Tool 1. I arrive and the only action that is needed is. 

ERCF_UNLOCK
ERCF_EJECT
T1
RESUME

And it works.

moggieuk, 2023-08-27:
There are lots of internal retries already (to the point of getting confusing).  That said, I get your point about a "top level" retry path.  I'll put this on the consideration list for the upcoming Happy Hare v2 replacement -- which is coded in a better way to accommodate such a retry..

moggieuk, 2023-09-04:
I have this on the "todo" list for Happy Hare v2