moggieuk/ERCF-Software-V3 #92: Feature Request: change endless spool mechanism

lettore, 2023-08-11:
Hello,
I set up the endless spool function but ùon my printer is not working as supposed. I have and Ender 3 S1 and my extruder is really low in height. This means that when the clog sensor feel that the filament is ended and try to unload the filament there's not enough filament in the extruder to reach the ERCF gear. So unload fail.
Is possible to use a clever approach and instead of unloading the previous filament just switch to the second filament and load it until it touch the filament already in the bowden tube.
So:
1 - try to move the selector to the next gate, if selector moves this means that filament is just finished and not clogged in the extruder
2 - as we know the distance that trigger the clog it's possible to slowly put the second filament in until it touches the previous filament in the bowden tube triggering a sensorless homing.
3 - restart the print and push the filament into the tube until it reach the distance to the extruder
4 - after that return to normal operation and bring anti clog on again

FlorianWilk, 2023-08-13:
Good thing you put the FeatureRequest here. I had posted mine on https://github.com/moggieuk/Happy-Hare/issues/9

moggieuk, 2023-08-27:
Yes, I do think this is possible (although harder than it seems because of slight calibration mismatch that is amplified over a large distance (the length of bowden tube).  It would require a high degree of calibration between extruder (moving original filament) and gear (moving new filament) as well as some logic to compensate.

All that to say it is one the "ideas" list but I have experimented yet...

FlorianWilk, 2023-08-27:
Pushing new Filament behind old one is btw a good way to do this calibration. 
Cut the filament before it goes into the ERCF, push the "new end" until it gets grip in the ERCF and then see if they have the a gap when both ends reach the Toolhead. Tune until you have a small gap at the toolhead (Extruder slightly faster than ERCF Gear) and then go to the previous settings without gap.

lettore, 2023-08-27:
Not sure why this should be challenging, I'm already using syncro extruding without any issue, so while printing both extruder and ercf are pushing the filament, it's just a matter to find a safe way to know if the filament is clogged or finished. Then the other problem is to sense when the new filament touch the old one, then continue printing with syncro enabled.