moggieuk/ERCF-Software-V3 #93: False trigger of "gate empty"

ssackett, 2023-08-18:
I had a filament break on a particular gate and fixed it.  After restarting the print the next time that gate was accessed I received a message that the gate was empty.  The gate was not automatically tested before the "gate empty" message was issued.  I manually executed the "gate test" macro and that fixed the problem.

moggieuk, 2023-08-27:
Well, this is by design (I tend to check all used gates before the print using my prusaslicer PR and passing list of gates to "CHECK_GATES", but I guess it does make some sense to automatically perform the gate check -- I'll add that to my TODO list..

moggieuk, 2023-09-04:
Hi, I've thought about this some more and I don't think it is right to retry....
There is a gate "unknown" state where an attempt will be made to load.  Once Happy Hare has concluded the gate is empty, if the user changes that fact, then `ERCF_CHECK_GATES GATE=xxx` should be run.  OR better still, the new filament is loaded with `ERCF_PRELOAD GATE=x` which will also update state.

Note that ERCF_ is now MMU_ with Happy Hare v2.

Based on this I'm going to close this issue.  Please re-open if you see errors in my logic.

closed 2023-09-04 by moggieuk

ssackett, 2023-09-04:
Moggieuk,

I agree with your conclusion.  Since I intervened in the operation of the software it should be my responsibility to inform the software about what I did.  

Steve