moggieuk/ERCF-Software-V3 #94: Resume Temperature Issue

Durahl, 2023-08-22:
Greetings!

I'm currently using an older `v1.3.0-24-g6bae537` where I noticed a problem when I tried to resume a Print after pausing it...

The Print was set to start the first layer @ 265°C which then got reduced to 240°C ( yea, -25°C )... 10min in the issue with the curled-up Overhang became apparent needing fixing so I paused the Print using KlipperScreen HHEd, trimmed down the offending material and then resumed the Print again using KlipperScreen but nothing happened...

I then tried resuming it from within the Fluidd UI and again nothing happened until I noticed the Hotend Temperature again rising to 265°C after which it continued needing manual intervention from me to reduce it back to 240°C again 🤔

[ASA Bulb Mount 52.1826g 51m4s.zip](https://github.com/moggieuk/ERCF-Software-V3/files/12412195/ASA.Bulb.Mount.52.1826g.51m4s.zip)
File was Sliced using the current OrcaSlicer 1.6.4

Any thoughts on this? 🤨

moggieuk, 2023-08-27:
Hmmm, I wonder what is significant about 240 degrees... is that what your `min_temp_extruder` is set to?

I know I JUST rewrote the extruder temp restoration logic in Happy Hare v2 (upcoming replacement release) so that it is much more straightforward to understand.  However, it is very strange that the temperature kept going up to 265.

When you say, paused the print, did you just run PAUSE or MMU_PAUSE?  (I ask because the current temp is recorded when MMU_PAUSE is called -- which is the same thing that occurs on MMU error).

Durahl, 2023-08-27:
My `min_temp_extruder` ( the one in the ercf_parameters.cfg ) is set to 200°C - I think it's the default?

It's been a few days but I _think_ I Paused the print using the Pause Button found on the KlipperScreen Print Status Screen, not the one on the HappyHare ERCF Screen 🤔 I could be wrong though since I was familiarizing myself with a new CHT 1.0mm Nozzle in Vase Mode hence the huge temperature gap - Good Bed Adhesion required the higher First Layer Temps but for the StealthBurner to be able to keep up even at 100% Part Cooling Fan I also needed to drastically reduce the Temps for the rest of the Print 🤨

moggieuk, 2023-08-28:
Good info.  It is conceivable (based on my recollection on how the old logic worked) that Happy Hare was holding onto the temp 265 from a previous print that had a "mmu pause" or error.   There was a bug (fixed in HHv2) that didn't clear this value.  So... if on the next print you paused with regular "pause" this value would not be overwritten and thus when RESUME is called, Happy Hare restores to 265.

Obviously it shouldn't do this but I can see that it would have been possible.   As I said this is fixed in the upcoming release because 'RESTORE' clears this value to prevent accidental reset.

Durahl, 2023-08-28:
Speaking of Updates, I've just updated everything:
[image]
And KlipperScreen HHEd still produces an error ( in Fluidd )?
[image]
I guess I still need to comment KS HHEd out in Moonraker? 🤨

Maybe something else you might provide an insight... Do you by any chance use **_OrcaSlicer_** with the ERCF? 🤔
If so, how do you solve the problem of OS not moving the Nozzle off the Model and onto the Purge Block for Tip Forming thus creating a Blob onto the Model?

I'm currently working around this by having it move the Nozzle into a Corner of the Build Plate whenever a Tool Change is being Called...
```
G1 X3 Y3 F24000
T[next_extruder]
```
...but that is a terrible solution especially with many Tool Changes happening on the first few layers as it'll keep pushing filament into the same spot resulting in a giant blob of molten filament. Unfortunately adding something like a Z2 to raise the Nozzle a bit ends with OrcaSlicer refusing to send the Sliced G-Code to the Printer ( I'm a simple man... Not a smart one 😑 )

moggieuk, 2023-08-29:
The KlipperScreen "warning" is because I deliberately don't call it KlipperScreen because I didn't want to offend the original author because it is a fork - I might change this though.   Usually there is a way to suppress the warning - In Mainsail, you can say "ignore" (this is documented in the README)

To move the nozzle on a tool change you are doing the right thing but it is probably better to put that in the ercf_software.cfg macro for Tx or CHANGE_TOOL..., although I've improved things a lot in HHv2 by having formal PRE_UNLOAD and POST_LOAD macro options.  I also have a built in optional z_hop.

moggieuk, 2023-09-02:
The original issue with temperature resuming has been substantially reworked in the new Happy Hare v2 software.  If you upgrade to that and still have a problem let me know because I believe such a problem should be impossible now.