moggieuk/ERCF-Software-V3 #96: MCU 'ercf' shutdown: Timer too close

ChGeorgi, 2023-08-30:
Send: ercf_calibrate
Recv: // Start the complete auto calibration...
Recv: // You are running an old calibration version.
Recv: // It is strongly recommended that you rerun 'ERCF_CALIBRATE_SINGLE TOOL=0' to generate updated calibration values
Recv: // Homing ERCF...
Recv: !! Unsure of filament position, recovering state...
Recv: // Unloading filament...
Recv: // Unloaded 47.3mm of filament
Recv: // Tool T0 enabled
Recv: // Calibrating reference tool T0
Recv: // Klipper state: Shutdown
Recv: // Finding extruder gear position (try #1 of 3)...
Recv: // MCU 'ercf' shutdown: Timer too close
Recv: // This often indicates the host computer is overloaded. Check
Recv: // for other processes consuming excessive CPU time, high swap
Recv: // usage, disk errors, overheating, unstable voltage, or
Recv: // similar system problems on the host computer.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'ercf' shutdown: Timer too close
Recv: // MCU 'ercf' shutdown: Timer too close
Recv: // This often indicates the host computer is overloaded. Check
Recv: // for other processes consuming excessive CPU time, high swap
Recv: // usage, disk errors, overheating, unstable voltage, or
Recv: // similar system problems on the host computer.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'ercf' shutdown: Timer too close
Recv: ok

moggieuk, 2023-09-02:
Thanks for the report, but Timer Too Close error is really Klipper's way of saying it is overwhelm.  If happens repeatedly there may be a way to work around it.   Has this happened to you more than this one occasion?  What version of Happy Hare are you running (can see in moonraker update manager in Mainsail/Fluuid)

ChGeorgi, 2023-09-02:
Thanks. I changed from raspberry 2B to Orange Pi 2 zero. Problem seems to be a too weak cpu.

moggieuk, 2023-09-04:
Ah, yes. As soon as you add features above the base Klipper, you can run into problems like this.  Usually it is in the communication (USB) processing rather than the core CPU.   Good news is the prices of rpi 3/4 is coming down again.

I'm going to close this issue. Re-open if you have a fairly reproducible case..

closed 2023-09-04 by moggieuk