moggieuk/ERCF-Software-V3 #97: Feature request: Allow using extruder gears as filament sensor

dflemstr, 2023-09-08:
I've been using this extruder design where I added a filament sensor that detects filament by the extruder gears being pushed apart: https://www.printables.com/model/487742-large-gear-clockwork2-ercf-filament-sensor-ecas-bo

However, this sometimes triggers this error: https://github.com/moggieuk/ERCF-Software-V3/blob/330126c413175b4c1b610036e124d911f782ec75/extras/ercf.py#L2126C64-L2126C64

I would like there to be a parameter that lets me change this behavior. Namely, when the ERCF feeds the filament to the filament gears, it is expected that they will sometimes be pushed apart and trigger the filament sensor that way.

For now I have fixed this locally by removing the code entirely.

moggieuk, 2024-02-05:
You are referencing a deprecated very old version of Happy Hare... Are you using Happy Hare v2.4.2?

It should support this just fine...

dflemstr, 2024-02-05:
This is a very old issue, sorry for keeping it open :)

closed 2024-02-05 by dflemstr